FPGAwars/apio-atom #21: Error on saving verilog

electrified, 2017-08-19:
On saving a verilog file, I get the following error pop up in Atom:

```
Error: spawn iverilog ENOENT
    at exports._errnoException (util.js:1026:11)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:193:32)
    at onErrorNT (internal/child_process.js:359:16)
    at _combinedTickCallback (internal/process/next_tick.js:74:11)
    at process._tickCallback (internal/process/next_tick.js:98:9)
```

[image]

I've done an ```apio install --all``` and iverilog is located at:

```/home/ed/.apio/packages/toolchain-iverilog/bin/iverilog```

Using the verify option from the Apio menu seems to run iverilog OK.

electrified, 2017-08-20:
I now see the problem is actually caused by linter-verilog, which expects iverilog to be on the path, and when installed by apio it isn't.

Jesus89, 2017-11-05:
I have created a fork of the `linter-verilog` called `apio-linter-verilog`: https://atom.io/packages/apio-linter-verilog.

Now the `apio-ide` uses this new linter package that points to the apio's iverilog.

closed 2017-11-05 by Jesus89